open-lv/air-guard #3: Plates (PCB) forma un noformējums

kasparsd, 2021-09-23:
Kāda varētu būt PCB kontūra un noformējums, lai tas liktos interesants un iederētos klasēs un birojos?

Svarīgi ņemt vērā ārējo izmēru (cena) un praktiskās iespējas tādu formu saražot pietiekami mazos kvantumos, lai pieļautu izmaiņas. Pirmajām partijām varētu izmantot https://jlcpcb.com vai https://www.pcbway.com

kasparsd, 2021-09-23:
Te ir jaunākais variants, kas ir mazliet draudzīgāks par vienkārši zaļu plati:

[image]

martinsp, 2021-09-25:
Vai nevajadzētu samainīt vietām bateriju turētāju un radošo laukumu? Domāju, vai komponentes uz plates netraucēs rokai, lai varētu uzzīmēt/uzrakstīt kaut ko radošajā laukumā.

kasparsd, 2022-01-04:
Taisām ciet, jo šis jau ir izdarīts!

closed 2022-01-04 by kasparsd