open-lv/air-guard #4: CO2 sensora enerģijas patēriņš un deep-sleep atbalsts

kasparsd, 2021-09-23:
Vai kādi no pieejamajiem sensoriem piedāvā zemu enerģijas patēriņu vai deep-sleep atbalstu, lai būtu iespējams visu iekārtu darbināt tikai no 2 gab AA baterijām vismaz vienu gadu?

festlv, 2021-09-23:
1 gada darbības laiks no 2xAA baterijām ir netriviāls uzdevums, kura risināšanai var nākties pabīdīt citus constraintus (gatavu ESP32 moduļu izmantošana, viegla montāža).

Kvalitatīvās 2xAA baterijās ir ap 6Wh enerģijas. Lai gadu darbinātu no šādām baterijām, vidējam patēriņam no baterijām jābūt ~210uA@3.3V.

Sensirion SCD41 pietuvojas šim (datasheet sola 400uA vidējo patēriņu mērot reizi 5 minūtēs), bet stockā pašlaik redzu 5 gab. ar nākamajām piegādēm decembrī.

kasparsd, 2021-09-24:
Paldies par izpēti @festlv!

Izskatās, ka konkrētajam TTGO T5 modelim ir vairākas lietas, kam jāpievērš uzmanību, lai dabūtu zemu enerģijas patēriņu pat pie ESP32 deep-sleep. Te ir labs pavediens https://github.com/esphome/issues/issues/1458

1. CP2104 UART uz USB čips, kam "enable" kāja ir pie USB 5V, tāpēc cerams, ka tas vispār netiek darbināts, kad darbojas no baterijas.

2. W25Q32FW flash čipam CS kāja esot atstāta gaisā, kas nozīmē, ka pie ESP32 deep sleep tas joprojām darbojas https://github.com/esphome/issues/issues/1458#issuecomment-813915856

   > TTgo has bad energy efficiency design, biggest easy fix is a pullup on w25q32 spiflash CS pin, without it spiflash does not go to sleep even when MCU is sleeping, this has a significant impact on sleep consumption. Removing all LEDs helps also.

3. E-paper draiverim ir jāpadod speciāla komanda gulēšanai, jo citādi tas arī paliek nomodā.

Šodien mēģināju to TTGO T5 ar esphome deep-sleep, bet pieslēgtu pie USB un dabūju 9mA@5V ar ieslēgtiem LEDiem. Ar wifi un ekrāna zīmēšanu patēriņš ir pie 50-70mA @ 5V.

Domāju, ka jāsāk apsvērt micropython, jo tad varam vairāk kontrolēt izmantotās bibliotēkas, driverus un arī pašu izejas kodu. Vienīgi tad mums vajag Micropython draiverus gan ekrānam (HINK E0213A22), gan sensoriem. Nezinu, vai ir kaut kas gatavs.

kasparsd, 2021-09-24:
Un vēl viena potenciālā problēma ar TTGO T5 ir iebūvētais TP4054 lādēšanas kontrolieris, kas atbalsta tikai litija jonu akumulatorus. Tas nozīmē, ka AA baterijas nevarēs slēgt pa taisno pie esošā BAT pieslēgvietas, jo tad, pieslēdzot USB 5V, tas gribēs lādēt AA vai AAA, kā li-ion.

Anrijs, 2021-09-25:
Notestēju TTGO T5 V2.3 pie precīzāka mēraparāta. Deep-sleep rādīja 1.0-1.5mA patēriņu. Visi LED izslēgti, tukšs kods. Tajā pašā laikā, mana custom plate tērē zem 100uA.

kasparsd, 2021-09-25:
Liels paldies @Anrijs! Vai būtu iespējams pārbaudīt vai pull-ups pie W25Q32FW CS kājas tiešām samazina šo vēl 10x kā daži raksta?

Interesanti, ka [jaunākajam TTGO T5 V2.3.1](http://www.lilygo.cn/prod_view.aspx?TypeId=50031&Id=1393&FId=t3:50031:3) viņi sola 300uA:

[image]

Ja 90% laika tiešām ir tie 300uA un pie e-paper zīmēšanas uzlec 50mA, tad vajadzētu būt OK, ne? Vienīgi jāmāk pamodināt iekārtu no deep-sleep, kad piesprauž USB kabeli programēšanai.

> Tajā pašā laikā, mana custom plate tērē zem 100uA.

Vai tas ir ar ESP32?

Kāds ir tavas domas, ko šajā gadījumā labāk darīt, ja uzdevums prasa pēc ekrāna un plaši pieejamiem moduļiem?

Anrijs, 2021-09-25:
Pieleiku 10k pullup (lielāku uz ātro neatradu). Tagad rāda 400-500uA. Mana plate ir ar ESP32 WROOM-32D moduli.
Ja neizmanto šo TTGO gatavo risinājumu, tad jāskatās kas vispār no e-ink ir pieejams saprātīgās cenās. Ir variants zīmēt plati uz kuras to ESP32 moduli un citas komponentes salodē pa tiešo.

kasparsd, 2021-09-25:
> Tagad rāda 400-500uA.

Šis jau ir diezgan labi, ne! Vismaz tagad izklausās ticami, ka tas jaunais variants V2.3.1 sola 300uA.

Ja liek klāt SCD41 sensoru (450uA), tad sanāk ap 1mA. Ja pieliek vēl e-paper refresh reizi 5 minūtēs (pie SDC41 lasījuma), tad gan jau salasīsies visi 2mA. Ar 2000mA bačām sanāk 1000h jeb mazliet pāri mēnesim. Man šķiet pieņemami, ka reizi mēnesī kāds piespraustu sensoru pie USB aku-bankas uzlādēties.

Ar MH-Z19b gan ir trakāk, jo tam vidējais patēriņš pie ieslēgta ir ap 17mA un iesilšanas ilgums 3min. Nezinu, cik daudz viņš spētu atdzist pa piecām minūtēm, bet kopumā pat ar 50% izslēgtu barošanu sanāk lielas problēmas. Un šī ir ilustratīva problēma visiem NDIR sensoriem. Bet toties tie ir pieejami.

Ja SAFs tiešām var nodrošināt 100 gab ar SCD41, tad varētu uztaisīt savu moduli ar through-hole kājām, jo pliks virsmas montāžas sensors maksā ap EUR 30 un spriežot pēc [Seeed moduļa](https://wiki.seeedstudio.com/Grove-CO2_%26_Temperature_%26_Humidity_Sensor-SCD41/), nekādi pasīvie elementi nav vajadzīgi.

Lai nepārvērstu šo par enerģijas patēriņa optimizācijas uzdevumu, varbūt jāsāk pieņemt, ka šāds moduļveidīgs dizains pieļauj tikai pāris dienas/nedēļas/mēnešus ar baterijām un primāro lietojumu jāplāno kā iekārtu, kas pieslēgta pie patstāvīgas barošanas.

ArtursGailis1995, 2021-09-25:
Pašam ir sataisīts energo-nedraudzīgs risinājums uz ESP32, LCD, BME280, DHT22, un priekš CO2 ir MHZ-19B sensors. Diez vai šim projektam der NDIR sensori. Dati pa WiFi sūtas uz InfluxDB serveri un attēlojas ar Grafana, lai varu redzēt arī tendences, ne tikai esošos mērījumus. Viss jau pāris gadus barojas no veca iPhone lādētāja, bet varbūt šis projekts pamudinās kaut ko optimizēt vai vismaz pamērīt patēriņu. 

Šāds tāds kods man ir arī Githubā, bet tur galīgi viss pa mežam, sen nav atjaunots.

kasparsd, 2021-09-25:
@ArtursGailis1995 Izskatās, ka Tev jau ir praktiski gatavs tas, ko mēģinām uzbūvēt :) Vai būtu iespējams padalīties ar bildi, kur Tev šī iekārta stāv telpā un kā izskatās?

Un smieklīgi, ka šodien tieši nolēmām atteikties no energo-efektīvā risinājuma, lai nebūtu jāoptimizē lietas, kas nav svarīgas tieši izglītības aspektam. Taisīsim slēdzamu pie patstāvīgas barošanas un tad ar visiem LEDiem un pīkstuļiem, lai jautrāk. Te ir sarakstes sākums https://gitter.im/open-lv/air-guard?at=614f51158065e87a8e027450

kasparsd, 2021-09-25:
Tā kā šodien izlēmām atteikties no bateriju barošanas, tad šo jautājumu pagaidām uzskatām par slēgtu. Ja būs vajadzība pēc bateriju darbināmas ierīces, tad arī meklēsim sensorus un komponentes, kas to atbalsta. Projekta MVP fāzē nebūtu prātīgi koncentrēties uz tehnisku optimizācijas uzdevumu, kad vēl nav saprastas kopējās prioritātes un vajadzības.

closed 2021-09-25 by kasparsd