open-lv/air-guard #6: Izvēlēties programēšanas platformu

kasparsd, 2021-09-24:
Tā kā šis ir izglītojošs projekts, tad vajadzīga tāda platforma, kuru ir viegli darbināt bez specializēta programnodrošinājuma.

Gan BBC Micro:bit, gan M5 Stack izmanto Micropython, jo tam ir sekojoši labumi:

- Ierīces atmiņā atrodas konstants firmware un lietotāja programnodrošinājums tiek definēts Python skriptos, kas glabājas kā "faili" flash atmiņā.
- Visas lietotāja programmas iespējams ielasīt, pieslēdzoties ar UART. Jebkuru programmu var rediģēt, kā arī pievienot jaunu.
- Pieejama REPL saskarsne, kas ļauj lietotājiem darbināt kontroliera funkcijas bez jauna programnodrošinājuma ielādes.
- REPL saskarsni var darbināt arī caur Web USB, izmantojot jebkuru tīmekļa pārlūku, kas to atbalsta.
- Pieejams programnodrošinājums visām operētājsistēmām, lai rediģētu un pievienotu jaunas programmas -- https://codewith.mu un https://thonny.org
- MicroPython ir Python, kas ir ļoti populāra un noderīga programēšanas valoda.

festlv, 2021-09-25:
Draiveru esamība/neesamība manuprāt nav iemesls, lai neizmantotu MicroPython: nepieciešamības gadījumā trūkstošais būs jāuzraksta, bet tas attiecas uz jebkuru platformu: ja projekts nebalstās uz izstrādes komandas iepriekš validētu hw/sw kombināciju, pastāv zināms risks, ka kaut kam nebūs draiveru, vai tajos nebūs realizēta vajadzīga fīča, vai būs kritiskas problēmas.

MicroPython izmantošanai profesionālā vidē es tik viegli nepiekristu, bet šāda projekta kontekstā nevaru iedomāties nevienu vērā ņemamu risku, kas izrietētu no MicroPython izmantošanas.

kasparsd, 2021-09-26:
Pēc diskusijas čatā un šeit ir nolemts izmantot MicroPython. Tas ir aprakstīts arī projekta tehniskajās vadlīnījās https://github.com/open-lv/air-guard/blob/92e274c4fb08f116cc48bd734bb726daccee3572/README.md#technical-design

closed 2021-09-26 by kasparsd