open-lv/air-guard #7: Materiālu saraksts (bill of materials) -- kontrolieri, sensori un barošana

kasparsd, 2021-09-25:
Provizoriskais materiālu saraksts:

## Vietas sensoriem

- [MH-Z19b](https://www.winsen-sensor.com/sensors/co2-sensor/mh-z19b.html) CO2 + temperatūras sensors.
- [SCD41](https://www.sensirion.com/en/environmental-sensors/carbon-dioxide-sensors/carbon-dioxide-sensor-scd4x/) CO2, temperatūras, mitruma un trokšņa sensors. Šis ir surface-mount, tāpēc vajadzīga pārejas plate vai arī atbalsts esošiem moduļiem.
- DHT22 temperatūras un mitruma sensors.

## Barošana

- Li-ion akumulators (varbūt lietots) vai 2 gab. AA vai 3 gab. AAA. Ja izmanto AA vai AAA baterijas, tad jāatslēdz baterijas lādēšanas funkcija, pievienojot micro-USB vadu, jo lādēšanas kontrolieris nemāk ne-li-ion akumulatorus.
- Li-ion vai AA/AAA turētājs through-hole.
- Vieta [dc-dc boost modulim](https://www.aliexpress.com/item/32800430445.html) uz 3.3V, ja tiek lietotas AA/AAA baterijas.

## Kontrolieris un ekrāns, variants A

- [EUR 19](https://www.aliexpress.com/item/32869729970.html) par [Lilygo TTGO T5 V2.3.1](http://www.lilygo.cn/prod_view.aspx?TypeId=50031&Id=1393&FId=t3:50031:3) ar DEPG0213BN e-paper draiveri. Komplektā ir Jst 2pin 1.25mm gals, ko pielodēt pie akumulatora.
- 2 gab 2.54 mm kājiņu papi, kur ievietot kontrolieri/ekrānu.

## Kontrolieris un ekrāns, variants B

- [Wemos S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html)
- [Wemos ePaper 2.13 Shield](https://www.wemos.cc/en/latest/d1_mini_shield/epd_2_13.html)

Anrijs, 2021-09-26:
MH-Z19B/C boost vajadzēs uz 5V

kasparsd, 2021-09-26:
Izskatās, ka ESP32-DevKitC [padod USB 5V uz EXT_5V kāju](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf), kad pieslēgts USB:

[image]

Vai pareizi saprotu?

Šis, protams, der tikai tad, ja tiek lietots patstāvīgs USB pieslēgums, par kuru vakar nolēmām čatā.

Anrijs, 2021-09-26:
Ja barošana paliek tikai no USB, tad var no tā EXT 5V pin barot

kasparsd, 2021-09-27:
ESP32 devkitc piegādātāju opcijas:

- https://www.olimex.com/Products/IoT/ESP32/ESP32-DevKit-LiPo/open-source-hardware
- https://www.tme.eu/lv/details/esp32-devkitc-32d/komplekts-datu-transmisijai/espressif/
- https://www.az-delivery.de/en/products/esp-32-dev-kit-c-v4?variant=30871551901792

kasparsd, 2021-09-27:
Esmu pasūtījis kopumā 15 gab. devkitc no diviem piegādātājiem un varēšu pārsūtīt visiem interesentiem.

kasparsd, 2021-10-09:
Iedomājos, ka varētu paredzēt vietu vēl sekojošām komponentēm:

- on/off slēdzis. Var likt pie `EN` kājas, bet tas izslēdz tikai 3.3V barošanu pašam ESP32 un visam, kas barojas no tiem 3.3V. Alternatīva ir paredzēt vietu savam USB virsmaz montāžas štepselim uz plates, kas var pilnībā izslēgt visu iekārtu. Pieslēgvieta uz plates arī atļautu optimizēt tās novietojumu, lai plate negāztos uz galda vai nesvērtos pie sienas.

- Passive buzzeris, kas nodzied bēdīgu meldiņu, kad pārsniedz 1400PPM un priecīgu, kad ir atpakaļ zem 600 vai jebkuras citas iedomātas robežas. Buzzerim vajadzētu savu slēdzi, lai var apklusināt bez pārprogramēšanas. Šiem parasti ir divas kājas, kuras var iespraust ligzdā.

- Kaut kādus ledus? Varbūt zaļu, dzeltenu un sarkanu vai vairākus rindā, lai atvieglotu CO2 nolasīšanu no attāluma, jo dzīvē tas SSD1306 ir tiešām maziņš un blāvs?

Anrijs, 2021-10-11:
`EN` varbūt labāk neaiztikt. Tas tiek izmantots priekš programmēšanas.
Buzeris varētu būt noderīgs, bet tam jātaisa driveris vai jāmeklē gatava plate ar driveri. Slēgt vienu kāju pa tiešo pie ESP32 nebūs labs variants.
LED varētu būt noderīgi, lai var redzēt stāvokli arī no otra telpas stūra.

closed 2022-01-20 by skrubis