open-lv/air-guard #10: Basic firmware in MicroPython that supports data display (OLED + 3 LEDs) and sensor calibration.

kasparsd, 2022-01-04:
Šis būtu pats minimums, ko vajadzētu nodrošināt, lai iekārta būtu funkcionāla (bet ne pārāk interesanta):

- Attēlo CO2 PPM uz ekrāna.
- LED uz rokas deg:
   - Zaļš, ja zem 700ppm
   - Dzeltens starp 700-1400ppm
   - Sarkans virs 1400ppm
- Pārsniedzot 1400ppm iepīkstas pīkstulis trīs reizes.
- Nospiežot rokas pogu ilgāk par 5 sekundēm tiek sākta kalibrēšanas procedūra.

kasparsd, 2022-01-04:
Bonus punkti par #12, kas piešķir iekārtai mākoņa spējas.

kasparsd, 2022-01-04:
Te ir prototips no @festlv https://wokwi.com/arduino/projects/319303888677110355

festlv, 2022-01-08:
Par CO2 sliekšņiem- varbūt pieturēties pie Aranet sliekšņiem (1000, 1000-1400, 1400+)?

Pašreizējais MicroPython todo:
* Pārliecināties, ka kāds neapstrādāts exception nerezultējas iefrīzojušā sargā,
* MHZ19C kontrolsummas rēķināšana ir kļūdaina, pieliku debug kodu, jāsagaida kad reproducēsies,
* Ieslēgt kalibrāciju (domāju ne tikai 5s nospiestu pogu, bet ar apstiprināšanu uz ekrāna)
* Ieslēdzoties detektēt sensora uzsilšanu (pēc novērojumiem, atgriež vērtību 500 daždesmit sekundes pēc starta, bet droši vien ir kāds statusa bits neoficiāli dokumentētajās komandās)
* Pīkstuļa iepīkstēšanās pārsniedzot 1400 (ar laika/mērījuma histerēzi)

festlv, 2022-01-11:
Vēl no TODO:
* Izslēgt automātisko nulles punkta kalibrāciju (saskaņā ar sensora dokumentāciju tā ir ieslēgta pēc noklusējuma)
* Build process kas ieraksta firmware versiju kādā .py failā
* Pīkstēšana pie kalibrācijas ieslēgšanas

kasparsd, 2022-01-11:
> Build process kas ieraksta firmware versiju kādā .py failā

Tu domā kaut kādu visa .py kopuma versiju? Varbūt iemest to kā pēdējo tajā `config` datnē?

To tad sūtītu ar MQTT ziņām, piemēram? Un rādītu uz ekrāna kaut kur?

festlv, 2022-01-11:
Jā, build procesa rezultāts būtu arhīvs ar py failiem, iespējams flasher skriptiem. Tas būtu piemērots brīdis arī vai nu config.py vai iekš sargs.py iemest __VERSION__ vai kāda nu bija pitona konvencija versijas glabāšanai. Pašu versiju esmu pieradis ņemt no git taga (ja commitam nav atbilstoša taga, tad pēdējais tags + commit hash.

Izmantot: parādīt uz ekrāna pēc ieslēgšanās, mqtt, kā arī ielogot pie startupa seriālajā konsolē.

closed 2022-02-17 by festlv