open-lv/air-guard #12: On-device webpage for configuring wifi credentials and remote data logging opt-in to our MQTT endpoint

kasparsd, 2022-01-04:
Ierīces web serveris ļauj ieslēgt (1) datu sūtīšanu un (2) attēlošanu Gaisa sarga “mākonī”.

Ja datu sūtīšana ieslēgta, tad iekārta reizi minūtē sūta MQTT ziņu uz `mqtt.gaisasargs.lv:1883`, kurā norāda:
MQTT servisa piekļuves dati:

- Lietotājs: iekārtas MACID ([a-z0-9]{12})
- Parole: Pirmie četri baiti no MicroPython `machine.unique_id()` HEX pierakstā ([a-z0-9]{8})

Sūtījuma dati:

- CO2 mērījums (int)
- Temperatūras mērījumu (int, jo sensors to dod neprecīzu)
- Publisko datu attēlošanas atļauja (boolean)

kasparsd, 2022-01-14:
@festlv Cik sarežģīti ir uztaisīt kaut ko no šī? Vai ir jau kādas gatavas bibliotēkas, kas nodrošina web serveri un kaut kādu konfigurācijas parametru kartēšanu uz mainīgo vērtībām?

festlv, 2022-01-14:
Pašlaik vēl neesmu pētījis, @RonnyLV izrādīja interesi palīdzēt ar šo uzdevumu.
Vienīgā piezīme, ka man izskatās, ka MicroPython machine.unique_id() atgriež to pašu MAC adresi.