open-lv/air-guard #25: Reduce OLED contrast on init

skrubis, 2022-02-10:
Due to incredible amounts of burn in caused by the somewhat static main screen of platformio fw running for maybe a day- please reduce the OLED contrast to a somewhat useable minimum on init

	i2c_command(0x81); // SSD1306_SETCONTRAST

festlv, 2022-02-14:
It seems to me that u8g2 library sets the contrast to maximum on startup.

badge.team's ssd1306 driver sets the contrast to 207 (range: 0-255). I would guess that this value is somewhat validated by badges, however, the real effect of this register's value also depends on bias resistor, which may differ (as the badges typically are using OLED directly, without the carrier board): https://forums.adafruit.com/viewtopic.php?t=38569#p573606

festlv, 2022-02-17:
After my screen was frozen for a day, it did experience burn-in with the previous contrast value (0xcf).
Reduced to 0x33 in https://github.com/open-lv/micropython/tree/0.0.8

closed 2022-02-17 by festlv