open-lv/air-guard #27: Append part of device MAC ID to the hotspot AP name to make them unique

kasparsd, 2022-02-28:
To avoid problems connecting to devices when multiple of them have hotspots enabled nearby we need to use unique AP names.

One solution is to append the last two bytes of the device MAC to the AP name such as `GaisaSargs03AB` or `GaisaSargs-03AB` here:

https://github.com/open-lv/air-guard/blob/bd06151d986bbe80923d18c140d2af601c506134/firmware/micropython/sargs.py#L291

kasparsd, 2022-02-28:
Can be done here https://github.com/open-lv/micropython/blob/39bf805c69e84e9c77e767989de3cdb7921e35e2/ports/esp32/boards/AIRGUARD/modules/network_manager.py or directly in `sargs.py`.

RonnyLV, 2022-02-28:
Fixed in https://github.com/open-lv/air-guard/commit/995cdb4d79f742ae6de5641e2b2abca483565a75

closed 2022-02-28 by RonnyLV