open-lv/air-guard #31: ERROR: jq not found! See https://stedolan.github.io/jq/

austrisv, 2022-04-29:
sekoju https://github.com/open-lv/air-guard/blob/main/tools/flasher/README.md instrukcijai;
palaidu requriements;
laižu flash.sh, man saka

`ESPTOOL_PORT=/dev/tty.usbserial-2110 ./flash.sh
ERROR: jq not found! See https://stedolan.github.io/jq/`

droši vien requirements.txt jāieliek jq=1.2.2

austrisv, 2022-04-29:
hmmm, jeb arī nē, ar pip3 install jq vēl ir par īsu?
`Successfully installed jq-1.2.2

% ESPTOOL_PORT=/dev/tty.usbserial-2110 ./flash.sh
ERROR: jq not found! See https://stedolan.github.io/jq/`

kasparsd, 2022-04-29:
Ar `jq` ir domāts šis https://stedolan.github.io/jq/, kas ir jāuzinstalē līdzīgi `esptools` pakotnei.

Te tiek skatīts, vai tāda programma sistēmā eksistē:

https://github.com/open-lv/air-guard/blob/d55d055b51f0e40992cd948609c789b28f72a6c5/tools/flasher/flash.sh#L23-L27

Reāli tā tiek izmantota šeit JSON atribūtu lasīšanai pie firmware lejuplādes:

https://github.com/open-lv/air-guard/blob/d55d055b51f0e40992cd948609c789b28f72a6c5/tools/flasher/flash.sh#L82-L83

austrisv, 2022-04-29:
ou yee, `brew install jq` did the magic. nu rūc uz nebēdu

kasparsd, 2022-04-29:
Forši! Varbūt vēlies papildināt to pirmo rindiņu "Run it on your local machine if you have both esptool.py and ampy installed or use the included Docker environment." ar `jq` atsauci un saiti blakus `ampy`? Citiem atvieglotu dzīvi :)

austrisv, 2022-04-29:
pamēģināju ... ļaušu ievērtēt, pielabot un/vai akceptēt :) paldies par atbalstu!